armbian/build #8232: [Bug]: Khadas oowow image generation with BTRFS is broken

lockheed, 2025-05-28:
### What happened?

The image is generated and seen in OOWOW as valid, but after flashing it by OOWOW, there's just bootloop with black screen. 

Here's the command line:

./compile.sh EXPERT="yes" ROOTFS_TYPE=btrfs BTRFS_COMPRESSION=zstd EXT="image-output-oowow" KHADAS_OOWOW_BOARD_ID=VIM1S

resulting image can be downloaded [here](https://share.impactor.one/s/4h6WIbwC).

I also tried generating an image without ROOTFS_TYPE=btrfs BTRFS_COMPRESSION=zstd and it booted fine.

### How to reproduce?

Generate an image with root on BTRFS.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-06-05:
> Maybe it's something to do with UBOOT not supporting BTRFS for boot drive, but the last time I've seen that claimed was 5+ years ago, and since they that should have been fixed.

VIM1S is not using mainstream u-boot and in such cases we need to lower expectations. Booting from BTRFS is probably not supported, u-boot is from 2019 ... This is pretty much corners case, which could cost a lot of time with resources we don't have and with unsure results. If u-boot is some [old fork](https://github.com/armbian/build/blob/main/config/sources/families/meson-s4t7.conf#L30), it was usually setup by one person once and even its open source, its not maintained by anyone else. Upgrade to modern mainstream is often mission impossible. It has to be done from scratch and can take years to completion. Until this hardware doesn't get that - we can't invest not even hours, while months are needed ... use what works.

closed 2025-06-05 by igorpecovnik

igorpecovnik, 2025-06-06:
> Choosing root as Btrfs results in Vim1S not booting at all, just stuck at blank screen.

Serial console logs are must have to debug this kind of operations. This method should in theory work and sometimes is perhaps easily fixable if logs are provided.

> Is it possible to network-boot on Vim1s?

Don't know - attach serial console / check on router if it gets IP address when no boot media is attached. This is rarely default enabled option, but some boards does support that. But how-to from here - again don't have an answer. Check forums and (official Khadas docs) for clues.