armbian/build #8261: [Bug]: Can't 'apt upgrade' to newer Kernel

scalpel4k, 2025-06-01:
### What happened?

Hi,

currently it seems impossible to update a running system to newer Kernels. armbian-bsp-* packages are only available on beta.armbian.com and not on apt.armbian.com.

However, the packages on beta.armbian.com have a hard dependency on the base-files package that isn't updated in any of the repos, at least not with a matching version.

Is there a chance to fix that? If I see this correctly it affects all systems and OSes.

thx

### How to reproduce?

On a running system try apt upgrade with beta.armbian.com in the sources, this will generate a conflict. Using apt.armbian.com instead will not show any kernel related updates.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-06-05:
tl;dr; Trixie is not yet part of stable repository, so packages are not generated. 

This configuration determines which packages will be build:
https://github.com/armbian/os/blob/main/userpatches/targets-release-standard-support.yaml

Maintaining all this codebase and system for generation represent significant efforts and in past few years I didn't get any notable help here, while I am totally overloaded. To prevent burnout, I limit my focus on stable userspace generation and supported hardware, while everything else is (auto)generated and works "as is". There Trixie should probably work. To make this work also in stable, several steps are required:
1. Adding codename (trixie) section to https://github.com/armbian/os/blob/main/userpatches/targets-release-standard-support.template#L30-L45
2. Changing variable build image to "no"
3. Regenerating stable artifacts and update repository

Armbian can't hire people to do this instead of me, so its up to you / anyone to do this. Once Trixie is out, perhaps not right away, I will run search and replace bookworm with Trixie in that file. Automation will do the rest.

ZoomRmc, 2025-06-07:
> In [nightly beta](https://github.com/armbian/os/blob/main/userpatches/targets-release-nightly.yaml#L192-L222) repository Trixie upgrade should work, but beta repository is not 100% stable.

Should it? Considering, userspace builds are excluded for `"focal", "sid", "trixie", "oracular", "plucky"`? Looks like this should break upgrades for rolling releases.

`beta` repos don't have `base-files` packages at all now, so it seems you're stuck with what you got with the image.

https://github.com/armbian/os/blob/20c9e2239259c76a6ba78ef4ed3e08d6cefa9da2/userpatches/targets-all-not-eos.yaml#L30

https://github.com/armbian/os/blob/20c9e2239259c76a6ba78ef4ed3e08d6cefa9da2/userpatches/targets-all-not-eos.yaml#L80

igorpecovnik, 2025-06-07:
OK, enabled, https://github.com/armbian/os/commit/302af69c4866cd6eda2a82a7a439c09e67a6c81f 

Lets see if this produces artifacts as expected.

igorpecovnik, 2025-06-08:
Beta repo now has correct base-files:
https://fi.mirror.armbian.de/beta/pool/trixie-utils/b/base-files/

Todo: update stable repo.

scalpel4k, 2025-06-09:
this has solved the issue indeed. thanks a lot!

closed 2025-06-16 by leggewie