armbian/build #8301: [Feature Request]: # Rockchip RK3588 octa core whatever fixme

CarlFK, 2025-06-15:
### Which feature would you like to have?

skimming sources, saw this, figured I should remind whoever 

https://github.com/armbian/build/blob/main/config/boards/orangepi5-ultra.csc#L1

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2025-06-15:
Devices labeled as `Community Supported` (.csc) are maintained on a best-effort basis by the community. Due to the large number of such devices, direct support from the core team is limited.

We gladly accept pull requests that add or improve configuration for these boards, as long as they do not break the build process. If a configuration causes build failures, it may be removed without prior notice to maintain project stability.

The same applies to cosmetic issues (e.g., board names, typos, formatting): they are only addressed if someone contributes a PR. Otherwise, they may remain uncorrected.

Thank you for your understanding and support!

EvilOlaf, 2025-06-16:
Ah that thing....PR pending

closed 2025-06-16 by EvilOlaf