armbian/build #8389: [Bug]: First login without internet: sed: -e expression #1, char 0: no previous regular expression

tosiara, 2025-07-17:
### What happened?

When booting for the first time without internet, `sed` error is printed on the welcome screen:

```
Generating locales: en_US.UTF-8
    _             _    _                                         _ _
   /_\  _ _ _ __ | |__(_)__ _ _ _    __ ___ _ __  _ __ _  _ _ _ (_) |_ _  _
  / _ \| '_| '  \| '_ \ / _` | ' \  / _/ _ \ '  \| '  \ || | ' \| |  _| || |
 /_/ \_\_| |_|_|_|_.__/_\__,_|_||_|_\__\___/_|_|_|_|_|_\_,_|_||_|_|\__|\_, |
                                 |___|                                 |__/
 v25.8 rolling for radxa cubie a5e running Armbian Linux 6.14.0-rc1-dev-sun55iw3

 Packages:     Debian stable (bookworm)
 Updates:      Kernel upgrade enabled and 2 packages available for upgrade
 Support:      for advanced users (rolling release)
sed: -e expression #1, char 0: no previous regular expression

 Performance:

 Load:         6%                Uptime:       17 min
 Memory usage: 7% of 1.93G
 CPU temp:     54°C                Usage of /:   15% of 6.7G

 Commands:

 Configuration : armbian-config
 Upgrade       : armbian-upgrade
 Monitoring    : htop
```

### How to reproduce?

1. Flash fresh Armbian
2. Bring down internet access
3. Log in for the first time
4. Error printed

When logging for the second time - no error

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leggewie, 2025-07-18:
confirmed in the [forum](https://forum.armbian.com/topic/53581-v258-rolling-for-radxa-cubie-a5e-6140-rc1-dev-sun55iw3-sed-e-expression-1-char-0-no-previous-regular-expression/)

twwn, 2025-07-19:
Using Jun 26's __Sid_vendor_server-kali__, for me this occurs even with internet access on my Radxa Zero 3W.

(via an ethernet USB-C adapter; Armbian will report a valid IP address)

igorpecovnik, 2025-07-20:
> Using Jun 26's Sid_vendor_server-kali

Debian Sid nor this hardware are supported, so this is more like an uncharted territory with unfinished software support. However this bug was noticed also elsewhere and I think we need to rewrite ` /etc/update-motd.d/10-armbian-header`

This error pops up if IP is not recognized.