armbian/build #8396: [Feature Request]: Add kernels selectable in armbian-config to armbian/build

AaronNGray, 2025-07-18:
### Which feature would you like to have?

Would it be possible to add selection of the older prebuilt kernels, kernel headers etc in `armbian-config` https://github.com/armbian/configng to be selectable in the EXPERT mode of build.

I tried looking at the `armbian-config` code and it looks quite complex with a sub module being loaded for the kernel functionality, otherwise would have done a pull request by now.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2025-07-19:
A couple of older kernel packages is available already within kernel change. We cannot keep all old packages forever since infrastructure costs money.

igorpecovnik, 2025-07-19:
>  Would it be possible to add selection

I understand that many ideas / requests may appear simple from the outside, but it’s important to recognize that for developers — especially those who rely on such work to support themselves — the challenge is not only technical complexity. Many feature requests, even seemingly minor ones, can require weeks or even months of effort and represent constant maintenance burden in the future. Who will be covering living expenses during this time is a serious and ongoing difficulty? If you target people that are bored, have plenty of time and have required technical understanding ... you might get luck and someone will take it from here.

We’ve already invested significant resources, we needed few years and we fund it with our private money, into developing `armbian-config` from scratch, which is now at a reasonably functional level. This foundation makes it technically feasible to implement features like the one you're suggesting — but many challenges remain. Already at the base level, which is not seen from this perspective.

The feature you're proposing is certainly useful and well-intentioned, but at the moment, we simply don’t have the resources to take it on (at least not this year) — if added, people will expect that it works. For us, every feature represents loosing more of private time and money plus raising maintaining expenses, which are far away from being covered by donations. Keep that in mind. Our priorities are focused on maintaining core functionality and supporting critical infrastructure with the limited capacity we have. That such ideas become possible.

It’s important to remember that once delivered, the feature, software, or service becomes freely available to everyone, including commercial competitors, there won't be any return of this investment.

If this feature is important to you, we encourage you to explore options such as:

- Investing your time / resources to implement it. Most of the features were done this way and this way you also come closer with the reality on how this is problematic - for everyone, including people that designed, developed and are maintaining this ecosystem.
- Finding someone willing to sponsor development, then find person to implement this for you.
- Starting a [crowdfunding campaign](https://forum.armbian.com/crowdfunding/) - convince users to share expenses and then find someone to implement this for you.
- keep [small donation](https://github.com/sponsors/armbian) so we can keep this functional

We’re happy to provide some support and guidance where we can, but ultimately, sustainable progress requires **shared** effort and initiative.

AaronNGray, 2025-07-19:
@igorpecovnik - okay I will futher look into developing this myself, also enquiring how much this development would cost as I dont have a lot of money. I am not the best bash programmer and find a lot of the code cryptic but I have isolated the armbian-config code and have it working under my own subdirectory. I will print out all the code as I work best on paper.

closed 2026-09-25 by EvilOlaf