armbian/build #8433: [Feature Request]: uefi_x86 oldlegacy 5.15

lodarmayarder, 2025-07-26:
### Which feature would you like to have?

I tried reintroducing kernel 5.15 during compilation. The latter notably supports proprietary NVIDIA graphics card drivers. I tried unsuccessfully by retrieving the file in config/kernel/linux-uefi-x86-legacy.config with the one from Armbian 23.11 and modifying config/sources/families/include/uefi_common.inc (legacy)

declare -g KERNEL_MAJOR_MINOR="5.15"
This compiled, but the image doesn't boot beyond grub. Is there a solution? May be create oldlegacy ? Thanks 

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2025-07-27:
> but the image doesn't boot beyond grub

A bite more detailed information about that might be helpful. You know, very few people have a second sight nowadays.

lodarmayarder, 2025-07-27:
Thanks and sorry for the lack of information. Do the changes I used seem consistent to you or do they risk weakening the build, could the option be implemented from the old-legacy base? During my first test, I forgot to replace the config/kernel/linux-uefi-x86-legacy.config file from Armbian 23.11 (5.15), I was using the one from the legacy kernel (6.1) and grub remained stuck on loading initial ramdisk. I have not yet tested the new compilation.