armbian/build #8436: [Bug]: Unable to build the Bullseye system: The repository 'http://deb.debian.org/debian bullseye-backports Release' does not have a Release file.

codesnas, 2025-07-28:
### What happened?

[🐳|🔨]   Types: deb
  [🐳|🔨]   URIs: http://security.debian.org/
  [🐳|🔨]   Suites: bullseye-security
  [🐳|🔨]   Components: main contrib non-free
  [🐳|🔨]   Signed-By: /usr/share/keyrings/debian-archive-keyring.gpg
  [🐳|🌱] Updating package list [ bullseye ]
  [🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-84144f59-5089-4f08-837c-e8c4afa99cd7/var/cache/apt'
  [🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-84144f59-5089-4f08-837c-e8c4afa99cd7/var/lib/apt/lists'
  [🐳|🔨]   Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
  [🐳|🔨]   Get:2 http://security.debian.org/ bullseye-security InRelease [27.2 kB]
  [🐳|🔨]   Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.0 kB]
  [🐳|🔨]   Ign:4 http://deb.debian.org/debian bullseye-backports InRelease
  [🐳|🔨]   Err:5 http://deb.debian.org/debian bullseye-backports Release
  [🐳|🔨]     404  Not Found [IP: 146.75.30.132 80]
  [🐳|🔨]   Get:6 http://github.armbian.com/configng stable InRelease [5,467 B]
  [🐳|🔨]   Get:7 http://deb.debian.org/debian bullseye/main arm64 Packages [7,955 kB]
  [🐳|🔨]   Get:8 http://deb.debian.org/debian bullseye/main Translation-en [6,235 kB]
  [🐳|🔨]   Get:9 http://deb.debian.org/debian bullseye/contrib arm64 Packages [40.8 kB]
  [🐳|🔨]   Get:10 http://deb.debian.org/debian bullseye/contrib Translation-en [46.9 kB]
  [🐳|🔨]   Get:11 http://deb.debian.org/debian bullseye/non-free arm64 Packages [72.4 kB]
  [🐳|🔨]   Get:12 http://deb.debian.org/debian bullseye/non-free Translation-en [92.5 kB]
  [🐳|🔨]   Get:13 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [16.3 kB]
  [🐳|🔨]   Get:14 http://deb.debian.org/debian bullseye-updates/main Translation-en [10.5 kB]
  [🐳|🔨]   Get:15 http://security.debian.org/ bullseye-security/main arm64 Packages [383 kB]
  [🐳|🔨]   Get:16 http://security.debian.org/ bullseye-security/main Translation-en [257 kB]
  [🐳|🔨]   Get:17 http://security.debian.org/ bullseye-security/contrib arm64 Packages [2,868 B]
  [🐳|🔨]   Get:18 http://security.debian.org/ bullseye-security/contrib Translation-en [2,512 B]
  [🐳|🔨]   Get:19 http://security.debian.org/ bullseye-security/non-free arm64 Packages [744 B]
  [🐳|🔨]   Get:20 http://security.debian.org/ bullseye-security/non-free Translation-en [1,208 B]
  [🐳|🔨]   Get:21 http://github.armbian.com/configng stable/main arm64 Packages [425 B]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://deb.debian.org/debian bullseye-backports Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🔨]   Hit:1 http://deb.debian.org/debian bullseye InRelease
  [🐳|🔨]   Hit:2 http://github.armbian.com/configng stable InRelease
  [🐳|🔨]   Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
  [🐳|🔨]   Ign:4 http://deb.debian.org/debian bullseye-backports InRelease
  [🐳|🔨]   Hit:5 http://security.debian.org/ bullseye-security InRelease
  [🐳|🔨]   Err:6 http://deb.debian.org/debian bullseye-backports Release
  [🐳|🔨]     404  Not Found [IP: 146.75.30.132 80]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://deb.debian.org/debian bullseye-backports Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🔨]   Hit:1 http://deb.debian.org/debian bullseye InRelease
  [🐳|🔨]   Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
  [🐳|🔨]   Ign:3 http://deb.debian.org/debian bullseye-backports InRelease
  [🐳|🔨]   Hit:4 http://security.debian.org/ bullseye-security InRelease
  [🐳|🔨]   Err:5 http://deb.debian.org/debian bullseye-backports Release
  [🐳|🔨]     404  Not Found [IP: 146.75.30.132 80]
  [🐳|🔨]   Hit:6 http://github.armbian.com/configng stable InRelease
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://deb.debian.org/debian bullseye-backports Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🚸] Command failed 3 times, giving up [ chroot_sdcard_apt_get_update ]
  Warning:  Command failed 3 times, giving up chroot_sdcard_apt_get_update
  [🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/rootfs/rootfs-create.sh:184
     create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:184
          create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
     artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:80
      artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
         obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
         build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:392
     get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
   ]
  Error:  Error 1 occurred in main shell at /armbian/lib/functions/rootfs/rootfs-create.sh:184
     create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:184
          create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
     artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:80
      artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
         obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
         build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:392
     get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50

### How to reproduce?

./compile.sh build BOARD=odroidn2 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no BUILD_ONLY=default COMPRESS_OUTPUTIMAGE=sha EXPERT=yes HOST=armbian KERNEL_CONFIGURE=no RELEASE=bullseye SHARE_LOG=yes

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/zihusehuve

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-07-28:
[🐳|🔨]   Reading package lists...
[🐳|🔨]   E: The repository 'http://deb.debian.org/debian bullseye-backports Release' does not have a Release file.

This is a problem on Debian side. Probably bullseye is not getting backports anymore. Uncomment backport line from Debian sources list file or upgrade to Bookworm.

igorpecovnik, 2025-07-29:
It somehow expected that backporting stops at certain moment. I don't know how this is actually managed on original Debian - do they remove the entry from the sources list or its just left as is, manifesting an error on update ... ?

EvilOlaf, 2025-07-29:
Should we even include backports by default?

Vector-Cross, 2025-09-28:
这个应该是以历史分支编译bullseye，在源码中找到 `lib/functions/rootfs/distro-specific.sh` 搜索 backports关键字，在 "$release" == "bullseye" 时注释掉backports的镜像源就好了
确保bullseye-backports在日志输出是不生效的状态，日志示例：
```
  [🐳|🔨]   -rw-r--r-- 1 root root 56156 Mar 18  2023 debian-archive-keyring.gpg
  [🐳|🔨]   -rw-r--r-- 1 root root 54031 Mar 18  2023 debian-archive-removed-keys.gpg
  [🐳|🔨]   total 4
  [🐳|🔨]   -rw-rw-r-- 1 root root 68 Sep 28 05:21 armbian.list
  [🐳|🔨]   deb http://deb.debian.org/debian bullseye main contrib non-free
  [🐳|🔨]   #deb-src http://deb.debian.org/debian bullseye main contrib non-free
  [🐳|🔨]   
  [🐳|🔨]   deb http://deb.debian.org/debian bullseye-updates main contrib non-free
  [🐳|🔨]   #deb-src http://deb.debian.org/debian bullseye-updates main contrib non-free
  [🐳|🔨]   
  [🐳|🔨]   #deb http://deb.debian.org/debian bullseye-backports main contrib non-free
  [🐳|🔨]   #deb-src http://deb.debian.org/debian bullseye-backports main contrib non-free
  [🐳|🔨]   
  [🐳|🔨]   deb http://security.debian.org/ bullseye-security main contrib non-free
  [🐳|🔨]   #deb-src http://security.debian.org/ bullseye-security main contrib non-free
  [🐳|🌱] Updating package list [ bullseye ]
```