armbian/build #8438: [Bug]: Unable to build bullseye: The repository 'http://security.debian.org bullseye/updates Release' does not have a Release file.

codesnas, 2025-07-29:
### What happened?

This modification is still incorrect: https://github.com/armbian/build/commit/cbeb0992b4c9061fb09776b1dbe45d729116d82e

```
  [🐳|🔨]   Types: deb
  [🐳|🔨]   URIs: http://deb.debian.org/debian
  [🐳|🔨]   Suites: bullseye bullseye-updates
  [🐳|����]   Components: main contrib non-free
  [🐳|🔨]   Signed-By: /usr/share/keyrings/debian-archive-keyring.gpg
  [🐳|🔨]   
  [🐳|🔨]   Types: deb
  [🐳|🔨]   URIs: http://security.debian.org/
  [🐳|🔨]   Suites: bullseye/updates
  [🐳|🔨]   Components: main contrib non-free
  [🐳|🔨]   Signed-By: /usr/share/keyrings/debian-archive-keyring.gpg
  [🐳|🌱] Updating package list [ bullseye ]
  [🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-6addef75-05a3-4125-9a19-d7ad33bac8b9/var/cache/apt'
  [🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-6addef75-05a3-4125-9a19-d7ad33bac8b9/var/lib/apt/lists'
  [🐳|🔨]   Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
  [🐳|🔨]   Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.0 kB]
  [🐳|🔨]   Get:3 http://github.armbian.com/configng stable InRelease [5,467 B]
  [🐳|🔨]   Ign:4 http://security.debian.org/ bullseye/updates InRelease
  [🐳|🔨]   Err:5 http://security.debian.org/ bullseye/updates Release
  [🐳|🔨]     404  Not Found [IP: 151.101.66.132 80]
  [🐳|🔨]   Get:6 http://deb.debian.org/debian bullseye/main arm64 Packages [7,955 kB]
  [🐳|🔨]   Get:7 http://deb.debian.org/debian bullseye/main Translation-en [6,235 kB]
  [🐳|🔨]   Get:8 http://deb.debian.org/debian bullseye/contrib arm64 Packages [40.8 kB]
  [🐳|🔨]   Get:9 http://deb.debian.org/debian bullseye/contrib Translation-en [46.9 kB]
  [🐳|🔨]   Get:10 http://deb.debian.org/debian bullseye/non-free arm64 Packages [72.4 kB]
  [🐳|🔨]   Get:11 http://deb.debian.org/debian bullseye/non-free Translation-en [92.5 kB]
  [🐳|🔨]   Get:12 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [16.3 kB]
  [🐳|🔨]   Get:13 http://deb.debian.org/debian bullseye-updates/main Translation-en [10.5 kB]
  [🐳|🔨]   Get:14 http://github.armbian.com/configng stable/main arm64 Packages [425 B]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://security.debian.org/ bullseye/updates Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🔨]   Hit:1 http://deb.debian.org/debian bullseye InRelease
  [🐳|🔨]   Hit:2 http://github.armbian.com/configng stable InRelease
  [🐳|🔨]   Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
  [🐳|🔨]   Ign:4 http://security.debian.org/ bullseye/updates InRelease
  [🐳|🔨]   Err:5 http://security.debian.org/ bullseye/updates Release
  [🐳|🔨]     404  Not Found [IP: 151.101.130.132 80]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://security.debian.org/ bullseye/updates Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🔨]   Hit:1 http://deb.debian.org/debian bullseye InRelease
  [🐳|🔨]   Hit:2 http://github.armbian.com/configng stable InRelease
  [🐳|🔨]   Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
  [🐳|🔨]   Ign:4 http://security.debian.org/ bullseye/updates InRelease
  [🐳|🔨]   Err:5 http://security.debian.org/ bullseye/updates Release
  [🐳|🔨]     404  Not Found [IP: 151.101.130.132 80]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   E: The repository 'http://security.debian.org/ bullseye/updates Release' does not have a Release file.
  [🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
  [🐳|🚸] Command failed 3 times, giving up [ chroot_sdcard_apt_get_update ]
  Warning:  Command failed 3 times, giving up chroot_sdcard_apt_get_update
  [🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/rootfs/rootfs-create.sh:184
     create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:184
          create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
     artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:80
      artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
         obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
         build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:392
     get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
   ]
  Error:  Error 1 occurred in main shell at /armbian/lib/functions/rootfs/rootfs-create.sh:184
     create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:184
          create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
     artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:80
      artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
         obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
         build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:392
     get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
```

### How to reproduce?

```
./compile.sh build RELEASE=bullseye BOARD=odroidn2 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no BUILD_ONLY=default COMPRESS_OUTPUTIMAGE=sha EXPERT=yes HOST=armbian KERNEL_CONFIGURE=no  SHARE_LOG=yes
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/runoxebobo

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-07-30:
> After an hour and a half of compilation, the error still indicates that the software source does not exist.

You mean rootfs was assembled from packages? 

Let me try full build.

igorpecovnik, 2025-07-30:
Tested. Now it should work. I recommend that you rebase your project to Bookworm base. Or even Trixie as its going to be next stable next month.

closed 2025-07-30 by igorpecovnik