armbian/build #8462: [Bug]: building Armbian with xfs filesystem not working

ArjanSchouten, 2025-08-05:
### What happened?

Building Armbian with xfs doesn't work with build framework version v25.5.1 and the latest trunk version v25.8.0-trunk.460.

Output:
```
[🐳|🌿] Enabling extension [ fs-xfs-support ]
modprobe: FATAL: Module xfs not found in directory /lib/modules/6.8.0-50-generic
[🐳|💥] error! [ Filesystem type unsupported by build host: xfs ]
[🐳|💥] Exiting with error 43 [ at /armbian/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
   check_filesystem_compatibility_on_host() --> lib/functions/configuration/main-config.sh:591
          do_main_configuration() --> lib/functions/configuration/main-config.sh:168
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
    do_with_conditional_logging() --> lib/functions/logging/section-logging.sh:111
    prep_conf_main_build_single() --> lib/functions/main/config-prepare.sh:29
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:22
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
```

### How to reproduce?

Build script:
```
export BOARD=orangepi5-max
export BRANCH=current
export RELEASE=bookworm
export BUILD_MINIMAL=yes
export KERNEL_CONFIGURE=no
export USE_CCACHE=yes
export KERNEL_BTF=no
export ROOTFS_TYPE=xfs

# Run Armbian compile script
./armbian-build/compile.sh \
    BOARD=${BOARD} \
    BRANCH=${BRANCH} \
    RELEASE=${RELEASE} \
    BUILD_MINIMAL=${BUILD_MINIMAL} \
    KERNEL_BTF=${KERNEL_BTF} \
    KERNEL_CONFIGURE=${KERNEL_CONFIGURE} \
    ROOTFS_TYPE=${ROOTFS_TYPE}
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-08-05:
Hi,

> modprobe: FATAL: Module xfs not found in directory /lib/modules/6.8.0-50-generic

All extension does, is adding the `xfsprogs` package which is looking for a module:
https://github.com/armbian/build/blob/main/extensions/fs-xfs-support.sh#L7C24-L7C32

I would say this is the problem of a host machine.

ArjanSchouten, 2025-08-06:
Thanks. I am on MacOS and that doesn't have support for XFS natively. A VM might help.

closed 2025-08-06 by ArjanSchouten