armbian/build #8468: [Bug]: Helios64 - ubuntu only - infinty loop on umount

BipBip1981, 2025-08-07:
### What happened?

Hi,
Only on ubuntu building, infinty loop on umounting.
Have a good day.

p64.last.manifest'
[🌿] Re-enabling [ initramfs-tools hook for kernel ]
[🔨]   mode of '/etc/kernel/postinst.d/initramfs-tools' changed from 0644 (rw-r--r--) to 0755 (rwxr-xr-x)
[🌱] Unmounting [ /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674 ]
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys: target is busy.
[🔨]   chsys /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys sysfs rw,relatime 0 0
[🔨]   efivarfs /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/dev/pts: no mount point specified.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/dev: not mounted
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/proc: not mounted.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys: target is busy.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/tmp: not mounted.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/var/tmp: not mounted.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/run/user/0: not mounted.
[🔨]   chsys /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys sysfs rw,relatime 0 0
[🔨]   efivarfs /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/dev/pts: no mount point specified.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/dev: not mounted
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/proc: not mounted.
umount: /home/armbianbuilder/build/.tmp/mount-0cf338f4-3c36-4528-ab5e-f6c104dbf674/sys: target is busy.

### How to reproduce?

git clone https://github.com/armbian/build ; cd build ; ./compile.sh
-> Do not change the kernel configuration
-> helios64
-> current
-> noble
-> image with console interface (server)
-> Minimal (Standard same but more time to reproduce, use Minimal ;-) )
 

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-08-07:
Hm 2nd attempt with debug enabled successful. https://paste.armbian.com/dekicufiqa

EvilOlaf, 2025-08-09:
Yeah got a couple of stuck Helios64 again, killed them.

prahal, 2025-08-10:
Seems not an issue on other boards because odroidm1, also a rockchip64 family, does not reproduce.
Could it be that before unmount,  after updating the initramfs in an image_from_sdcard_rootfs qemu chroot, we would trigger systemd-shutdown ?
Because helios64 has systemd-shutdown hook to mangle the gpio via /sys. Could it be it hangs in qemu and prevents unmounting the qemu chroot /sys?

EvilOlaf, 2025-08-11:
I could mitiage by adding `"sys/firmware/efi/efivars"` here:  https://github.com/armbian/build/blob/2d9cce1846b51d12216f812472324c9ce52542df/lib/functions/general/chroot-helpers.sh#L71

EvilOlaf, 2025-08-11:
I had the same question in mind but couldn't come to a conclusion. Just for fun I tried building the kernel without EFI support (yes, it has EFI by default, no clue why) but did not change anything on the issue.

EvilOlaf, 2025-08-13:
@BipBip1981 feel free to do tests against the PR mentioned above.