armbian/build #8482: [Bug]: What is the difference in bug and bugfix label?

leggewie, 2025-08-11:
### What happened?

main/.github/labels.yml

We have too many labels, I think.  What is the difference in bug and bugfix label, please?  Can't we make do with just 1?

### How to reproduce?

.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-08-11:
My guess is "bugfix" is for a pr that fixes a bug while "bug" is for an issue report reporting a bug

leggewie, 2025-08-21:
Thank you for the clarification.

Wouldn't it make sense to simplify that to a single label "bug"?

bug label on PR: it's a bugfix
bug label on an issue: it reports a bug

bugfix label on an issue: nonsense but currently possible

Comments?

leggewie, 2025-09-04:
Apparently the Armbian Administration prefers complexity over clarity and purpose and thinks this quirkiness adds value to Armbian.

Arm - your quirky distro - bian