armbian/build #8520: [Bug]: three-digit kernel patches are applied before two-digit ones

belegdol, 2025-08-19:
### What happened?

I tried to build 6.6.102 kernel for odroid XU4. Patching failed, because the 100→101 and 101→102 ended up being applied before others.

### How to reproduce?

1. Clone https://github.com/belegdol/build/tree/odroidxu4-6.6.102
2. `./compile.sh BOARD=odroidxu4 BRANCH=current RELEASE=bookworm KERNEL_CONFIGURE=no KEEP_KERNEL_CONFIG=no kernel`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leggewie, 2025-08-19:
Thank you for reporting, but I cannot verify the problem.

```
$ ./compile.sh BOARD=odroidxu4 BRANCH=current RELEASE=bookworm KERNEL_CONFIGURE=no KEEP_KERNEL_CONFIG=no kernel &>/dev/null && echo success
success
```

leggewie, 2025-08-19:
are you sharing your github repo?

$ git clone git@github.com:belegdol/build.git
Cloning into 'build'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

belegdol, 2025-08-19:
While not a bug in armbian per se, it currently prevents me from submitting an update to the odroid kernel. I tried fixing it myself but I was not able to get the python patcher to enter the debugger.

igorpecovnik, 2025-08-19:
Try with series file - where you control patch execution order.

leggewie, 2025-08-19:
why not simply fix it by renaming?  I'm happy to submit a PR.  Would it be sufficient to pad the double-digit files with a 0?

patch-6.6.95-96.patch -> patch-6.6.95-096.patch

PR #8522 (2025-08-19): pad double digits in odroidxu4-6.6 patch with a leading zero (#8520)
fixes #8520 for the odroidxu4-6.6 as well as the spacemit-6.6 kernel
closed 2025-08-19 by igorpecovnik