armbian/build #8571: Freeze v25.05 branch

sashasimkin, 2025-09-02:
### What happened?

Hi!
I’m using v25.05 branch to build images, but then the OS version turns out as v25.8-trunk
At the same time I see that sources are frozen on 25.05, but there are differences with 25.5.1 tag
I need v25.05 for some rpi fixes which seem to be missing on 25.5.1

Do you have plans to freeze v25.5 or it’ll be skipped altogether?

Will you accept a pull request to fix the version?

(I tried updating version to v25.05-{commit-hash} but that resulted in v25.11-trunk)

### How to reproduce?

Build armbian with v25.05

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-09-02:
Fixed with https://github.com/armbian/build/commit/50cecd999713136a9b039a76172e9fadf91afaab

closed 2025-09-02 by igorpecovnik

sashasimkin, 2025-09-02:
Thank you very much! Appreciate rapid reaction!